mysensors/MySensors #1538: High GPIO pin number doesn't work when compiling mysgw

DocRuben, 2023-01-02:
I'm trying to compile mysgw with the IRQ pin set to 360 wich is the pin number for my SBC(Orange Pi 3 LTS)
but after compiling the number has been changed to 104 and that causes an error at runtime.

> ERROR attachInterrupt: Unable to open GPIO edge interface for pin 104: Permission denied

Works fine without specifying an IRQ pin, probably because all other pin-numbers are < 255

mfalkvidd, 2023-01-03:
[attachInterrupt](https://github.com/mysensors/MySensors/blob/253109d3ff00ec524c5d1e1dfcd8e197c96e54c2/hal/architecture/Linux/drivers/core/interrupt.cpp#L140) takes an uint8_t, so any value above 255 will result in whatever C does when things don't fit (360 - 256 is 104 so I guess it just ignores the higher bits). This normally result in warnings during compilation. Could you post the full output of the compilation?

mfalkvidd, 2023-01-03:
Could you also provide the full configure command? That would make it possible to try to reproduce the problem.

mfalkvidd, 2023-01-03:
When using `./configure --my-rf24-irq-pin=360` I get the following warnings, confirming that 360 is truncated to 104:
```
./hal/transport/RF24/driver/RF24.cpp: In function ‘bool RF24_initialize()’:
<command-line>: warning: unsigned conversion from ‘int’ to ‘uint8_t’ {aka ‘unsigned char’} changes value from ‘360’ to ‘104’ [-Woverflow]
./hal/transport/RF24/driver/RF24.cpp:522:12: note: in expansion of macro ‘MY_RF24_IRQ_PIN’
  hwPinMode(MY_RF24_IRQ_PIN,INPUT);
            ^~~~~~~~~~~~~~~
In file included from ./MySensors.h:39,
                 from examples_linux/mysgw.cpp:82:
<command-line>: warning: unsigned conversion from ‘int’ to ‘uint8_t’ {aka ‘unsigned char’} changes value from ‘360’ to ‘104’ [-Woverflow]
./hal/architecture/Linux/drivers/core/Arduino.h:40:62: note: in definition of macro ‘digitalPinToInterrupt’
 #define digitalPinToInterrupt(pin) RPi.digitalPinToInterrupt(pin)
                                                              ^~~
./hal/transport/RF24/driver/RF24.cpp:537:48: note: in expansion of macro ‘MY_RF24_IRQ_PIN’
  RF24_SPI.usingInterrupt(digitalPinToInterrupt(MY_RF24_IRQ_PIN));
                                                ^~~~~~~~~~~~~~~
<command-line>: warning: unsigned conversion from ‘int’ to ‘uint8_t’ {aka ‘unsigned char’} changes value from ‘360’ to ‘104’ [-Woverflow]
./hal/architecture/Linux/drivers/core/Arduino.h:40:62: note: in definition of macro ‘digitalPinToInterrupt’
 #define digitalPinToInterrupt(pin) RPi.digitalPinToInterrupt(pin)
                                                              ^~~
./hal/transport/RF24/driver/RF24.cpp:539:40: note: in expansion of macro ‘MY_RF24_IRQ_PIN’
  attachInterrupt(digitalPinToInterrupt(MY_RF24_IRQ_PIN), RF24_irqHandler, FALLING);
                                        ^~~~~~~~~~~~~~~
./hal/architecture/Linux/MyMainLinuxGeneric.cpp: In function ‘void handle_sigint(int)’:
<command-line>: warning: unsigned conversion from ‘int’ to ‘uint8_t’ {aka ‘unsigned char’} changes value from ‘360’ to ‘104’ [-Woverflow]
./hal/architecture/Linux/MyMainLinuxGeneric.cpp:45:18: note: in expansion of macro ‘MY_RF24_IRQ_PIN’
  detachInterrupt(MY_RF24_IRQ_PIN);
```

DocRuben, 2023-01-03:
My complete configure command getting this error is
> ./configure --spi-spidev-device=/dev/spidev1.0 --my-transport=rf24 --my-rf24-channel=76 --my-rf24-ce-pin=117 --my-rf24-cs-pin=227 --my-rf24-irq-pin=360 --my-rf24-pa-level=RF24_PA_MAX

When compiling I get a lot of warnings about different vars getting truncated both with and without specifying a IRQ-pin, in both cases it compiles sucsessfully but with the IRQ-pin specified I get the previous error at runtime.
Without it compiles and seams to run fine eaven though all warnings.

Attaching my output from make.
[make.txt](https://github.com/mysensors/MySensors/files/10338894/make.txt)

chey, 2024-08-11:
I get this as well. This is due to the changes in the Linux kernel related to gpio on sysfs and the offset needed to turn on the correct pins.

Related to #1561.

Maybe the easy solution here would be to allow for pin numbers greater than 255.

chey, 2024-08-15:
> I get this as well. This is due to the changes in the Linux kernel related to gpio on sysfs and the offset needed to turn on the correct pins.
> 
> Related to #1561 and #1536.
> 
> Maybe the easy solution here would be to allow for pin numbers greater than 255.

At least in these locations `uint8_t` should be changed to `uint16_t`: 
https://github.com/mysensors/MySensors/blob/e298769eb73ba2da781a34d66cae345c6840b7c0/hal/architecture/Linux/drivers/core/interrupt.cpp#L140
https://github.com/mysensors/MySensors/blob/e298769eb73ba2da781a34d66cae345c6840b7c0/hal/architecture/Linux/drivers/core/interrupt.cpp#L140