mysensors/MySensors #1551: Mysensors RFM69 gateway compilation error on 64-bit OS (RFM69_MAX_PACKET_LEN)

frepkovsky, 2023-08-26:
Just for tracking this issue that has already been [discussed in the forum](https://forum.mysensors.org/topic/11456/is-mysensors-rpi-gw-32bit-only/12) long time ago but is not fixed yet.

When trying to build RFM69 mysgw on 64-bit OS the following error occurs:

```
test@raspberrypi:~/MySensors $ make
g++ -MT build/examples_linux/mysgw.o -MMD -MP -march=armv8-a+crc -mtune=cortex-a53 -DMY_RADIO_RFM69 -DMY_RFM69_NEW_DRIVER -DMY_GATEWAY_LINUX -DMY_DEBUG -DLINUX_SPI_BCM -DLINUX_ARCH_RASPBERRYPI -DMY_RFM69_FREQUENCY=RFM69_868MHZ -DMY_IS_RFM69HW -DMY_PORT=5003  -Ofast -g -Wall -Wextra -std=c++11 -I. -I./core -I./hal/architecture/Linux/drivers/core -I./hal/architecture/Linux/drivers/BCM -c examples_linux/mysgw.cpp -o build/examples_linux/mysgw.o
In file included from ./MySensors.h:388,
                 from examples_linux/mysgw.cpp:82:
./hal/transport/RFM69/driver/new/RFM69_new.cpp: In function ‘void RFM69_interruptHandling()’:
./hal/transport/RFM69/driver/new/RFM69_new.cpp:282:62: error: no matching function for call to ‘min(long unsigned int, unsigned int)’
  282 |                                          RFM69_MAX_PACKET_LEN);
      |                                                              ^

```

Workaround for this problem on 64-bit OS is to change RFM69_MAX_PACKET_LEN definition in **RFM69_new.h**:

```
test@raspberrypi:~/MySensors $ diff ./hal/transport/RFM69/driver/new/RFM69_new.h.orig ./hal/transport/RFM69/driver/new/RFM69_new.h
130c130
< #define RFM69_MAX_PACKET_LEN             (0x40u)              //!< This is the maximum number of bytes that can be carried 
---
> #define RFM69_MAX_PACKET_LEN             (0x40ul)             //!< This is the maximum number of bytes that can be carried
```

and then compilation works fine. 

But this fix/workaround cannot be simply committed like above in the sources because doing it this way solves the problem on 64-bit OS but cause failure on 32-bit OS. It has to be done some other way...

frepkovsky, 2023-08-30:
This fix works for me on 64-bit and 32-bit OS - I tested building it and also running gateway and it works fine. But I'm not cpp programmer so better it is reviewed...

```diff
diff --git a/hal/transport/RFM69/driver/new/RFM69_new.cpp b/hal/transport/RFM69/driver/new/RFM69_new.cpp
index ed1f48b4..5357b6f3 100644
--- a/hal/transport/RFM69/driver/new/RFM69_new.cpp
+++ b/hal/transport/RFM69/driver/new/RFM69_new.cpp
@@ -278,7 +278,8 @@ LOCAL void RFM69_interruptHandling(void)
                                     RFM69.currentPacket.header.packetLen - 1);
 
                        if (RFM69.currentPacket.header.version >= RFM69_MIN_PACKET_HEADER_VERSION) {
-                               RFM69.currentPacket.payloadLen = min(RFM69.currentPacket.header.packetLen - (RFM69_HEADER_LEN - 1),
+                               RFM69.currentPacket.payloadLen = min(static_cast<uint>
+                                                                    (RFM69.currentPacket.header.packetLen - (RFM69_HEADER_LEN - 1)),
                                                                     RFM69_MAX_PACKET_LEN);
                                RFM69.ackReceived = RFM69_getACKReceived(RFM69.currentPacket.header.controlFlags);
                                RFM69.dataReceived = !RFM69.ackReceived;

```

Givy, 2023-12-08:
Hi Frepkovsky,
I am having the same issue, see below.  Do you have your fix steps documented?  I would like to follow your fix.  Thank you so much.

/MySensors# make
gcc -MT build/hal/architecture/Linux/drivers/core/config.o -MMD -MP -march=armv8-a+crc -mtune=cortex-a53 -mfpu=neon-fp-armv8 -mfloat-abi=hard -DMY_RADIO_RF24 -DMY_GATEWAY_LINUX -DMY_DEBUG -DLINUX_SPI_BCM -DLINUX_ARCH_RASPBERRYPI  -Ofast -g -Wall -Wextra  -I. -I./core -I./hal/architecture/Linux/drivers/core -I./hal/architecture/Linux/drivers/BCM -c hal/architecture/Linux/drivers/core/config.c -o build/hal/architecture/Linux/drivers/core/config.o
**_gcc: error: unrecognized command-line option ‘-mfpu=neon-fp-armv8’
gcc: error: unrecognized command-line option ‘-mfloat-abi=hard’_**
make: *** [Makefile:103: build/hal/architecture/Linux/drivers/core/config.o] Error 1

PR #1559 (2024-01-28): Fix building RFM69 gw on 64-bit OS
Building mysgw with RFM69 driver fails with error on 64-bit OS because there were 2 fixes #1500 and #1551 for the problem (with RFM69_MAX_PACKET_LEN) merged to development branch. Both patches fix the issue by other way and both works when applied separately but building gw is not possible when both are applied... 

I did rollback my fix #1551 to keep the fix #1500 only.
closed 2024-01-28 by tekka007