mysensors/MySensors #1555: Documentation change for Arduino pro micro pinout for serial gateway

heltrix, 2023-11-14:
When setting up an NRF24 serial gateway it fails on my Arduino pro micro unless the CE and CSN pins are manually specified by adding the lines:

#define MY_RF24_CE_PIN 9
#define MY_RF24_CS_PIN 10

The following debug logs are shown.

0;255;3;0;9;40371 TSM:FAIL:DIS
0;255;3;0;9;40371 TSF:TDI:TSL
0;255;3;0;9;50373 TSM:FAIL:RE-INIT
0;255;3;0;9;50373 TSM:INIT
0;255;3;0;9;50378 !TSM:INIT:TSP FAIL
0;255;3;0;9;50379 TSM:FAIL:CNT=4
0;255;3;0;9;50379 TSM:FAIL:DIS
0;255;3;0;9;50379 TSF:TDI:TSL

mfalkvidd, 2023-11-14:
@heltrix what board are you using (in the Arduino IDE)? What version of the board definitions?
Selecting a Pro Micro board such as the Lilypad sets `ARDUINO_ARCH_AVR` which automatically selects the correct pins at https://github.com/mysensors/MySensors/blob/0af72015da44e689d6cd9ab6afe4e0f6ecb01d16/hal/transport/RF24/driver/RF24.h#L69 but that doesn't seem to be the case for the setup you are using.

mfalkvidd, 2023-11-14:
If I add
```
#define XSTR(x) STR(x)
#pragma message "CE pin used: " XSTR(DEFAULT_RF24_CE_PIN)
#pragma message "CS pin used: " XSTR(DEFAULT_RF24_CS_PIN)
```
after `#include <MySensors.h>` in https://github.com/mysensors/MySensors/blob/development/examples/GatewaySerial/GatewaySerial.ino#L90 I get the following output:
```
GatewaySerial.ino:93:57: note: #pragma message: CE pin used: (9)
GatewaySerial.ino:94:57: note: #pragma message: CS pin used: (SS)
```