mysensors/MySensors #1561: RPi interrupt management not working on latest Linux version (6.6)

romaricr, 2024-03-11:
I have upgrade my Raspberry OS to version 6.6 of Linux. After that, my MySensors gateway was not working anymore displaying an error related to interrupt management.
After some review, it appears that sysfs management of interrupts is obsolete - see [here](https://www.[kernel.org/doc/Documentation/gpio/sysfs.txt](https://www.kernel.org/doc/Documentation/gpio/sysfs.txt)) and [there](https://www.thegoodpenguin.co.uk/blog/stop-using-sys-class-gpio-its-deprecated/).
Newest versions should use `gpiod.h` headers and `libgpiod ` as documented [here](https://www.ics.com/blog/gpio-programming-exploring-libgpiod-library) for example.

For the moment, disabling interrupts is working.

romaricr, 2024-03-11:
I realise that my post is related to issue #1536. I had interesting references though. Don't hesitate to merge if necessary.

romaricr, 2024-03-12:
I have a fix now. Once I get some time I will post the results

chey, 2024-08-10:
> I have a fix now. Once I get some time I will post the results

@romaricr How did you make out? Happy to test your fix.