mysensors/MySensors #1563: Compile error with ESP32-S2 Feather and MySensor Example Sketch

wint0178, 2024-04-16:
I am having a compile error in Arduino IDE v.2.3.2 installed on my MacBook Air running OSX 14.4.1 when I attempt to verify or upload an example sketch from the MySensors Library (v.2.3.2) onto an ESP32-S2 Feather with BME280 board. The sketch is the “GatewayESP32” example from the library with no changes made. I have gone through and checked to see if I have any duplicate libraries and removed some that appeared to be listed in the compile error and then reinstalled, but none of this has worked. I also uninstalled and reinstalled the ESP32 board. Thanks in advance if you are able to help figure this out.
[Compile Error.txt](https://github.com/mysensors/MySensors/files/15000980/Compile.Error.txt)
[GatewayESP32.txt](https://github.com/mysensors/MySensors/files/15000982/GatewayESP32.txt)

mfalkvidd, 2024-04-16:
The esp32 library version (2.0.15) is probably incompatible. Try downgrading to ESP32 1.0.4.

wint0178, 2024-04-17:
This appears to have removed the "Adafruit Feather ESP32-S2"  from the available boards in in the manager and the only one that is now available and seems relevant is the "ESP32 Dev Module", but I am getting an error with that one also. I copied the last few lines of the compile error below, but only the last two lines were in red starting with "exec:" 

Using library MySensors at version 2.3.2 in folder: /Users/bjw58994/Documents/Arduino/libraries/MySensors 
Using library WiFi at version 1.0 in folder: /Users/bjw58994/Library/Arduino15/packages/esp32/hardware/esp32/1.0.4/libraries/WiFi 
Using library EEPROM at version 1.0.3 in folder: /Users/bjw58994/Library/Arduino15/packages/esp32/hardware/esp32/1.0.4/libraries/EEPROM 
Using library SPI at version 1.0 in folder: /Users/bjw58994/Library/Arduino15/packages/esp32/hardware/esp32/1.0.4/libraries/SPI 
exec: "python": executable file not found in $PATH

Compilation error: exec: "python": executable file not found in $PATH

wint0178, 2024-04-17:
The lines above what I shared previously include the only other mention of the word "python" in the entire error. I have them copied below. I am also planning to do some searching for this error online and will post an update if I find a solution.

Archiving built core (caching) in: /private/var/folders/jj/xqz4b04n3sdc0p0591lk9wr067pzwj/T/arduino/cores/5256a81cfda5790d3a36fd47107ee112/core.a
Linking everything together...
/Users/bjw58994/Library/Arduino15/packages/esp32/tools/xtensa-esp32-elf-gcc/1.22.0-80-g6c4433a-5.2.0/bin/xtensa-esp32-elf-gcc -nostdlib -L/Users/bjw58994/Library/Arduino15/packages/esp32/hardware/esp32/1.0.4/tools/sdk/lib -L/Users/bjw58994/Library/Arduino15/packages/esp32/hardware/esp32/1.0.4/tools/sdk/ld -T esp32_out.ld -T esp32.common.ld -T esp32.rom.ld -T esp32.peripherals.ld -T esp32.rom.libgcc.ld -T esp32.rom.spiram_incompatible_fns.ld -u ld_include_panic_highint_hdl -u call_user_start_cpu0 -Wl,--gc-sections -Wl,-static -Wl,--undefined=uxTopUsedPriority -u __cxa_guard_dummy -u __cxx_fatal_exception -Wl,--start-group /private/var/folders/jj/xqz4b04n3sdc0p0591lk9wr067pzwj/T/arduino/sketches/E6B6CC2A50D4C824C8AEFD761E19BB2D/sketch/GatewayESP32.ino.cpp.o /private/var/folders/jj/xqz4b04n3sdc0p0591lk9wr067pzwj/T/arduino/sketches/E6B6CC2A50D4C824C8AEFD761E19BB2D/libraries/MySensors/MyASM.S.o /private/var/folders/jj/xqz4b04n3sdc0p0591lk9wr067pzwj/T/arduino/sketches/E6B6CC2A50D4C824C8AEFD761E19BB2D/libraries/WiFi/ETH.cpp.o /private/var/folders/jj/xqz4b04n3sdc0p0591lk9wr067pzwj/T/arduino/sketches/E6B6CC2A50D4C824C8AEFD761E19BB2D/libraries/WiFi/WiFi.cpp.o /private/var/folders/jj/xqz4b04n3sdc0p0591lk9wr067pzwj/T/arduino/sketches/E6B6CC2A50D4C824C8AEFD761E19BB2D/libraries/WiFi/WiFiAP.cpp.o /private/var/folders/jj/xqz4b04n3sdc0p0591lk9wr067pzwj/T/arduino/sketches/E6B6CC2A50D4C824C8AEFD761E19BB2D/libraries/WiFi/WiFiClient.cpp.o /private/var/folders/jj/xqz4b04n3sdc0p0591lk9wr067pzwj/T/arduino/sketches/E6B6CC2A50D4C824C8AEFD761E19BB2D/libraries/WiFi/WiFiGeneric.cpp.o /private/var/folders/jj/xqz4b04n3sdc0p0591lk9wr067pzwj/T/arduino/sketches/E6B6CC2A50D4C824C8AEFD761E19BB2D/libraries/WiFi/WiFiMulti.cpp.o /private/var/folders/jj/xqz4b04n3sdc0p0591lk9wr067pzwj/T/arduino/sketches/E6B6CC2A50D4C824C8AEFD761E19BB2D/libraries/WiFi/WiFiSTA.cpp.o /private/var/folders/jj/xqz4b04n3sdc0p0591lk9wr067pzwj/T/arduino/sketches/E6B6CC2A50D4C824C8AEFD761E19BB2D/libraries/WiFi/WiFiScan.cpp.o /private/var/folders/jj/xqz4b04n3sdc0p0591lk9wr067pzwj/T/arduino/sketches/E6B6CC2A50D4C824C8AEFD761E19BB2D/libraries/WiFi/WiFiServer.cpp.o /private/var/folders/jj/xqz4b04n3sdc0p0591lk9wr067pzwj/T/arduino/sketches/E6B6CC2A50D4C824C8AEFD761E19BB2D/libraries/WiFi/WiFiUdp.cpp.o /private/var/folders/jj/xqz4b04n3sdc0p0591lk9wr067pzwj/T/arduino/sketches/E6B6CC2A50D4C824C8AEFD761E19BB2D/libraries/EEPROM/EEPROM.cpp.o /private/var/folders/jj/xqz4b04n3sdc0p0591lk9wr067pzwj/T/arduino/sketches/E6B6CC2A50D4C824C8AEFD761E19BB2D/libraries/SPI/SPI.cpp.o /private/var/folders/jj/xqz4b04n3sdc0p0591lk9wr067pzwj/T/arduino/sketches/E6B6CC2A50D4C824C8AEFD761E19BB2D/core/core.a -lgcc -lesp32 -lphy -lesp_http_client -lmbedtls -lrtc -lesp_http_server -lbtdm_app -lspiffs -lbootloader_support -lmdns -lnvs_flash -lfatfs -lpp -lnet80211 -ljsmn -lface_detection -llibsodium -lvfs -ldl_lib -llog -lfreertos -lcxx -lsmartconfig_ack -lxtensa-debug-module -lheap -ltcpip_adapter -lmqtt -lulp -lfd -lfb_gfx -lnghttp -lprotocomm -lsmartconfig -lm -lethernet -limage_util -lc_nano -lsoc -ltcp_transport -lc -lmicro-ecc -lface_recognition -ljson -lwpa_supplicant -lmesh -lesp_https_ota -lwpa2 -lexpat -llwip -lwear_levelling -lapp_update -ldriver -lbt -lespnow -lcoap -lasio -lnewlib -lconsole -lapp_trace -lesp32-camera -lhal -lprotobuf-c -lsdmmc -lcore -lpthread -lcoexist -lfreemodbus -lspi_flash -lesp-tls -lwpa -lwifi_provisioning -lwps -lesp_adc_cal -lesp_event -lopenssl -lesp_ringbuf -lfr -lstdc++ -Wl,--end-group -Wl,-EL -o /private/var/folders/jj/xqz4b04n3sdc0p0591lk9wr067pzwj/T/arduino/sketches/E6B6CC2A50D4C824C8AEFD761E19BB2D/GatewayESP32.ino.elf
python /Users/bjw58994/Library/Arduino15/packages/esp32/hardware/esp32/1.0.4/tools/gen_esp32part.py -q /Users/bjw58994/Library/Arduino15/packages/esp32/hardware/esp32/1.0.4/tools/partitions/default.csv /private/var/folders/jj/xqz4b04n3sdc0p0591lk9wr067pzwj/T/arduino/sketches/E6B6CC2A50D4C824C8AEFD761E19BB2D/GatewayESP32.ino.partitions.bin