mysensors/MySensors #1564: How many of you are using PJON?

gioblu, 2024-04-18:
Hi all, I am not sure this the right place to ask this question. I am the creator of the PJON Protocol, I took some time out from the project, but I am really curious how many of you effectively made good use of it and if you are satisfied or encountered any issue with it.

paultbarrett, 2024-05-07:
@gioblu , I have been using PJON for my house for over 2 years and it's great. All my light switches are on this. They have pushbutton switches and PIRs. These all go back to MySensors and trigger automations in Home Assistant.

I have noticed that there wasn't anything done on PJON for 2 years now. Is this still something being worked on ?

gioblu, 2024-05-24:
Ciao @paultbarrett yes I have stopped to actively develop the implementation. There are still some bugs to be fixed, in particular we have some issues with ESPs and SWBB strategy, I hope I will have some time to be back working on PJON soon.