mysensors/MySensors #1567: Configure script not picking the CPU type

raks-dev, 2024-10-21:
I was setting up MQTT gateway on my ```Raspberry Pi 4 Model B Rev 1.1``` running on Homebridge image ```6.6.51+rpt-rpi-v8``` which is a 64 Bit version.
Build would fail after configure as the flags ```-mfpu=neon-fp-armv8``` and ```-mfloat-abi=hard``` get added to it.

I was able to root cause the issue to the below mentioned code:
```
if [ -z "${SOC}" ]; then
    printf "${SECTION} Detecting target machine.\n"
    info=($(detect_machine))
    SOC=${info[0]}
    TYPE=${info[1]}
    CPU=${info[2]}
    printf "  ${OK} machine detected: SoC=${SOC}, Type=${TYPE}, CPU=${CPU}.\n"
fi

if [ -z "${CPUFLAGS}" ]; then
    CPUFLAGS=$(gcc_cpu_flags "${SOC}" "${CPU}")
fi
```
https://github.com/mysensors/MySensors/blob/development/configure#L575-L586

The CPU variable is set only when SOC flag is unset but used in determining gcc_cpu_flags and results in following error

```
gcc: error: unrecognized command-line option ‘-mfpu=neon-fp-armv8’
gcc: error: unrecognized command-line option ‘-mfloat-abi=hard’
```

Proposed Fix:
Add the line ```CPU=$(eval "uname -m 2>/dev/null")``` before the ```if [ -z "${CPUFLAGS}" ]; then``` and deleting     ```CPU=${info[2]}``` line.

cvdenzen, 2025-01-30:
Here are my results of the patched configure. It doesn't recognise the CPU.
`
x8@rpi4:~/project/MySensors $ ./configure
[SECTION] Detecting target machine.
  [OK] machine detected: SoC=unknown, Type=unknown, CPU=.
[SECTION] Checking GPIO Sysfs.
  [OK] /sys/class/gpio/export found.
[SECTION] Detecting SPI driver.
  [OK] SPI driver detected:SPIDEV.
[SECTION] Gateway configuration.
  [OK] Type: ethernet.
  [OK] Transport: rf24.
  [OK] Signing: Disabled.
  [OK] Encryption: Disabled.
  [OK] CPPFLAGS:  -DMY_RADIO_RF24 -DMY_GATEWAY_LINUX -DMY_DEBUG -DLINUX_SPI_SPIDEV 
  [OK] CXXFLAGS:  -std=c++11
[SECTION] Detecting init system.
  [OK] Init system detected: systemd.
[SECTION] Saving configuration.
  [OK] Saved.
[SECTION] Cleaning previous builds.
  [OK] Finished.
x8@rpi4:~/project/MySensors $ uname -a
Linux rpi4 6.6.62+rpt-rpi-v8 #1 SMP PREEMPT Debian 1:6.6.62-1+rpt1 (2024-11-25) aarch64 GNU/Linux
`
The patched lines:
`x8@rpi4:~/project/MySensors $ git diff configure
diff --git a/configure b/configure
index 1281ebe3..0b2139b2 100755
--- a/configure
+++ b/configure
@@ -577,10 +577,13 @@ if [ -z "${SOC}" ]; then
     info=($(detect_machine))
     SOC=${info[0]}
     TYPE=${info[1]}
-    CPU=${info[2]}
+    # 20250130 Carl CPU=${info[2]}
     printf "  ${OK} machine detected: SoC=${SOC}, Type=${TYPE}, CPU=${CPU}.\n"
 fi
 
+# 20250130 Carl https://github.com/mysensors/MySensors/issues/1567
+CPU=$(eval "uname -m 2>/dev/null")
+
 if [ -z "${CPUFLAGS}" ]; then
     CPUFLAGS=$(gcc_cpu_flags "${SOC}" "${CPU}")
 fi
`

cvdenzen, 2025-02-04:
For me it doesn't work. What branch did you start with (master, development)?
```
8@rpi4:~/project/MySensors $ uname -a
Linux rpi4 6.6.62+rpt-rpi-v8 #1 SMP PREEMPT Debian 1:6.6.62-1+rpt1 (2024-11-25) aarch64 GNU/Linux
```

jpolitow, 2025-02-07:
Solution for my was to add manually into build params 
soc and cpu-flags

for mine rasp pi 3 it was:
./configure --my-transport=rf24 --my-rf24-irq-pin=22 --soc=BCM2837 --cpu-flags="-march=armv8-a+crc -mtune=cortex-a53"

when you know your soc you can check cpu-flags here: https://github.com/mysensors/MySensors/blob/development/configure#L263-L305