mysensors/MySensors #1568: RF24 (nrf24l01) Radio Failures when Dynamic Payloads are used

TMRh20, 2025-05-04:
To keep this short, the line
https://github.com/mysensors/MySensors/blob/e298769eb73ba2da781a34d66cae345c6840b7c0/hal/transport/RF24/driver/RF24.cpp#L351
needs to be changed to
```cpp
    if(result > 32 || !result) {
}
```

Per the RF24 Datasheet:
> Note: Always check if the packet width reported is 32 bytes or shorter when using the
R_RX_PL_WID command. If its width is longer than 32 bytes then the packet contains errors
and must be discarded. Discard the packet by using the Flush_RX command.

It turns out this is also this case when it returns 0.

We discovered this just recently in the RF24 library itself. The available() function kept returning true, while dynamic payload size was 0. A radio read had no effect on clearing the buffer.

If users are noticing any random/long-term stability issues, this is the cause. It is very difficult to reproduce, can take minutes, hours, days, months, etc to happen in a production environment.

mfalkvidd, 2025-05-04:
Thanks for notifying us @TMRh20 !

closed 2025-05-05 by tekka007

TMRh20, 2025-05-09:
Just FYI I've opened an issue with Nordic:
https://devzone.nordicsemi.com/f/nordic-q-a/121237/nrf24l01-radio-r_rx_pl_wid-returns-0

PR #1569 (2025-05-05): Update RF24.cpp
Fix #1568
commit 3020658f (2025-05-05): Update RF24.cpp (#1569)

Fix #1568